Allen-Synthesis/EuroPi #4: [Project Suggestion] Improve oled initialization

awonak, 2021-11-24:
# Project Suggestion

Improve "[OSError: [Errno 5] EIO](https://github.com/Allen-Synthesis/EuroPi/blob/main/troubleshooting.md#oserror-errno-5-eio
)" error handling with more user friendly message.

**Describe your suggestion or idea**  

The troubleshooting guide provides helpful tips for handling "[OSError: [Errno 5] EIO](https://github.com/Allen-Synthesis/EuroPi/blob/main/troubleshooting.md#oserror-errno-5-eio
)". This error message can be improved by checking the I2C ready state before allowing the exception to be thrown.

In [europi.py](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/firmware/europi.py#L11-L13) lines 11-13, the oled I2C device is registered and immediately used. In the MicroPython documentation, the I2C class offers a "[scan](https://docs.micropython.org/en/latest/library/machine.I2C.html#machine.I2C.scan)" method which allows you to scan for peripheral deices on the given bus. If no devices are found, you can gracefully report that error to the user instead of allowing the unintuitive "OSError: EIO" error to be thrown.

An example implementation would look like this:

```python
oled_bus = I2C(0, sda=Pin(0), scl=Pin(1), freq=400000)
if len(oled_bus.scan()) == 0:
    raise SystemError("No I2C devices found. Ensure OLED device properly installed.")

oled = SSD1306_I2C(WIDTH, HEIGHT, oled_bus)
oled.fill(0)
oled.show()
```

**Images**  
If applicable, add screenshots or photos to help explain your problem.

roryjamesallen, 2021-11-26:
Excellent suggestion. MicroPython does not seem to support SystemError without extra packages being installed, so an alternative method of displaying error messages is needed.
I have used simple red text to indicate the error and then exited the program. It's not ideal but it does also work when the library is imported, showing the correct error message in whichever is the currently running program.
[image]

awonak, 2021-11-26:
+1 to not adding additional imports. This looks like a great solution to me.

closed 2021-11-28 by roryjamesallen