Allen-Synthesis/EuroPi #5: [Project Suggestion] Simplify OLED display extra funtions

awonak, 2021-11-24:
# Project Suggestion

Instead of having separate functions to add custom behavior to the OLED instance variable, you can subclass the SSD1306_I2C library and add your custom functions there.

**Describe your suggestion or idea**  

For example:
```python
class EuroPiDisplay(SSD1306_I2C):
    def __init__(self, sda: int, scl: int, channel: int = 0, freq: int = FREQ):
        i2c = I2C(channel, sda=Pin(sda), scl=Pin(scl), freq=freq)
        if len(i2c.scan()) == 0:
            raise SystemError("No I2C devices found. Ensure OLED device properly installed.")
            return
        
        # super() calls the init method on the SSD1306_I2C base class
        super().__init__(WIDTH, HEIGHT, i2c)
    
    def centre_text(self, text):
        """Takes a string of up to 3 lines separated by '\n', and displays them centred vertically and horizontally."""
        ...
        self.text(...)
    
    def clear(self):
        self.fill(0)

# Sample usage:
oled = EuroPiDisplay(SDA_PIN, SCL_PIN)
oled.clear()
oled.show()
# default SSD1306_I2C method
oled.text("default text")
# custom method
oled.centre_text("centred text")
```

This will give users a consistent interface for working with the OLED display. Custom functions should still be documented as such in the [OLED Tips page](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/oled_tips.md#extra-functions-from-europipy), but this will give users a consistent interface when coding up their scripts.

**Images**  
If applicable, add screenshots or photos to help explain your problem.

roryjamesallen, 2021-11-26:
Including the error message approach from [Issue 4](https://github.com/Allen-Synthesis/EuroPi/issues/4), the extra OLED functions are now accessed as methods, using a subclass named 'oled_display'.
This should also make it much easier for users to add their own custom OLED methods.

[image]

awonak, 2021-11-26:
One minor tweak to that change on [europi.py line #62](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/firmware/europi.py#L62), instead of `oled.text(...)` that should be `self.text(...)` just like you've done with self.clear() because self is the `oled_display` class instance and we no longer need to rely on the global variable.

roryjamesallen, 2021-11-28:
Fixed to use self.text() rather than oled.text(). A custom error message has now also been added for the even that a non-string is sent to the method

closed 2021-11-28 by roryjamesallen