Allen-Synthesis/EuroPi #6: [Project Suggestion] Add support for alternative OLED pin configurations

roryjamesallen, 2021-11-26:
# Project Suggestion

**Describe your suggestion or idea**  
The current PCBs only support one pin layout for the I2C bus of the OLED display, meaning that alternative displays using the same driver which may be more accessible to other customers cannot be used.
The PCB could contain some hardware to allow the pins to be switched for alternative layouts.

**Images**  
The Pi Hut OLED
[image]

CPC OLED (theoretically compatible but alternate pin configuration)
[image]

roryjamesallen, 2021-11-26:
Currently pursuing two possible ways of addressing this issue; hardware switches to allow configuration when the module is assembled, and a software + hardware approach where the pins can be switched using signals from GPIOs. The second would use up more pins of the Pico, but would result in much less hardware clutter and a much easier process for a DIY user.

roryjamesallen, 2021-11-26:
Further research shows only two common pin layouts exist for this OLED display, so a simpler approach is possible: A Hardware DIP switch and pull-up/down resistors.

#### The two potential pin layouts
[image]

#### Proposed hardware switching solution
[image]

In mode 1 (The Pi Hut), the DIP switch would be set to on/on/off/off
In mode 2 (CPC, AliExpress), it would be off/off/on/on

This would allow the two different OLED configurations to be used easily, with only a quick change of the switch positions during assembly to allow it to work.

#### Potential issues with solution
In theory, due to user error or misunderstanding, incorrect signals could be sent to the OLED, however this should not be an issue as the I2C bus runs on 3.3V, so even if the power rail were connected to a data pin, no damage should be caused and the issue could be fixed with a simple switch change.

roryjamesallen, 2021-11-26:
#### Illustration of proposed solution
[image]
DIP switch is situated underneath the OLED display to prevent accidental knocking of the switches during removal of the module, but switches would still be accessible with a pencil if required

awonak, 2021-11-26:
Another option I've seen modules use is jumpers and shunts to change hardware configuration. However, this probably wouldn't fit on the jack PCB and would have to be on the pico PBC.

Here's an example of a module I recently build that uses jumpers and shunts (regretfully I don't have the schematics). Check out page 65:
https://zlobmodular.com/wp-content/uploads/2021/01/svf_6612_builddocs-compressed.pdf

roryjamesallen, 2021-11-26:
> Another option I've seen modules use is jumpers and shunts to change hardware configuration. However, this probably wouldn't fit on the jack PCB and would have to be on the pico PBC.

Yup, I did consider using shunts and in many ways it would have been a cleaner approach, however there just isn't room on either PCB. The jack PCB doesn't have the vertical space between boards for a full jumper + shunt, and the pico PCB doesn't have the room around the existing traces and components

roryjamesallen, 2021-11-27:
A sheepish reply to my above comment: There may in fact be a much simpler header and shunt solution which would fit, as the shunts are shorter than I previously calculated, allowing for them to be mounted on the jack PCB but facing towards the pico PCB.  
  
As the pins alternate in a certain way between the two configurations, a solution with only an 8 way 2 row header and 4 shunts should be possible, taking up minimal space and being easy to set for the user:  
  
[image]

roryjamesallen, 2021-12-11:
Implemented solution works on new PCBs, users can even make the bridge simply using solder, saving the cost of shunts/jumpers if appearance isn't the priority!

[image]  
  
[image]

closed 2021-12-11 by roryjamesallen