Allen-Synthesis/EuroPi #8: [Project Suggestion] Front access USB and reset

roryjamesallen, 2021-11-28:
# Project Suggestion

**Describe your suggestion or idea**  
Currently the module needs to be plugged in via the USB on the Pico, but this means sometimes cramming the cable in between your rails, and you don't have any easy access to either bootsel or reset without removing the module.

It would be nice to have front access USB and potentially buttons for reset and bootsel.

roryjamesallen, 2021-11-28:
A solution is in progress which would use an expander module of 4HP.
The module would have front facing USB B (standard for Eurorack devices such as the Lich, Sol, or USB Pal), as well as the same buttons used on the EuroPi for bootsel and reset. It also includes a copy of the LED found on the Pico itself (GPIO25) to allow it to be used as a general purpose LED, perhaps for indicating digital input, or the current status of a variable in the current program

[image]
[image]
[image]

The expander would house the actual Pico, and the Pico on the actual EuroPi module would be replaced by a 'spoof Pico', a simple PCB with holes for headers, which would in turn connect the two with a 20 way wire.

This solution also has spare space to include headers for further expansion using the unused pins of the Pico.

roryjamesallen, 2021-11-29:
Further views of the concept:

Pico Side             |  Back Side
:-------------------------:|:-------------------------:
[image]  |  [image]

All unused pins are now accessible from the expansion headers at the bottom of the module, and dimensions are now correct

mjaskula, 2022-01-19:
I love everything about this, it should make working with the Europi as well as future expansion much easier. 

One thing that does worry me is the size of the buttons. The look like they might be easy to hit by accident. But I guess I don't know how bad that would that be.

roryjamesallen, 2022-01-19:
That's actually a very good point that completely slipped my mind. I had been in the mindset of wanting to make everything as big and spaced out as possible but making these buttons smaller would actually be much better for this. I'll look into smaller ones similar to those used for the quality selector on Mutable Beads

mjaskula, 2022-01-19:
Yeah, that sounds like the right size to me. 

Of course it’ll open up more panel real estate, which will make me want for more jacks and knobs.

roryjamesallen, 2022-01-19:
I reckon that's a good call! I might make it an alternate panel that has extra jacks/knobs as I really want this to not take away any I/O that people might want to use for their own expanders instead. It wouldn't be too hard to have an alternate version of the expander that adds a couple of extra ins and outs if people want them

roryjamesallen, 2022-01-22:
As suggested by @mjaskula, the current design now uses smaller tactile switches for the bootsel and reset functions to try and prevent accidental presses
[image]

noises-modular, 2022-06-30:
Could the spoof have a 2x10 pin header to allow for a common cable (assuming they are made)?

roryjamesallen, 2022-07-01:
That is actually a great idea; even better (more standard) would be a 16-16 pin Eurorack cable! Looking back at which pins would be broken out to the expander, it would be:
10 GPIO
+5V USB
GND
And then potentially some extra signals back from the Pico, such as regulated 3.3V

Anyway, the point is that fewer than 16 pins are used, so using a standard Eurorack cable would not only be more mechanically secure than a dupont style cable, but it's also much more likely to be easy to source for DIYers.

mjaskula, 2022-07-01:
One problem I see here with using the standard eurorack power cable is that someone might plug in a standard eurorack power cable and fry something.

roryjamesallen, 2022-07-01:
That is a very good point. There should be quite a lot of spare room around the connector if it is the full size of a Pico though, so perhaps with enough silkscreen warning it could be reliable?

It should also be relatively obvious as there will still be the normal power connector on the EuroPi, so hopefully it would seems illogical to connect two separate power cables to one module. I can also try to design the expander so that the 16 pin headers line up perfectly when the module and expander are side by side, so it should be pretty intuitive to connect a short 16-16 pin cable between the two, maybe if they're both labelled EXPANDER in obvious silkscreen too

awonak, 2022-07-01:
Identical pin headers on a PCB are definitely a concern, but there are many examples of modules that take this risk and provide detailed documentation to address this, for example the Zlob Entropy Expander uses 10 pin headers for its expander: https://zlobmodular.com/wp-content/uploads/2020/06/connecting_the_expander.pdf

noises-modular, 2022-07-02:
Another thought is I made a midi device with an arduino which uses a 3.5mm TRS type connection and communicates over serial to a sampler module in my rack. This expander may be a good opportunity to add an old school midi input and output to the Europi.

roryjamesallen, 2022-07-02:
That's definitely something I'll keep in mind, probably for a separate expander though as I'd rather leave as much up to the user as possible, and they might not want 2 of the GPIOs being used for MIDI if they'll never use it! It should be a pretty easy expansion module to make though, especially once the spare pins are available from the back of the USB expander

noises-modular, 2022-07-03:
I wonder if the Pico W might alleviate the need for USB access for re-programming.

Another thought on the expansion is to simplify the number of custom boards needed: a future rev of the Europi board could have a 2x16 shrouded pin header placed underneath where the Pico sits and the builder would have the option to populate it or not.

awonak, 2022-09-19:
One other potential benefit with a design like this is that you can have the microcontroller board with 2x20 female pin headers for standard breadboard sized microcontrollers like Arduino rp2040 Connect, or a Teensy 4.0. 

Connectors should ideally be a common layout that supports DIY with standard cables we might have lying around. Either two 2x10 (unshrouded to allow us to use 16>10 pin eurorack power cables leaving some of the 16 pin side unconnected) or one 2x20 which is common for classic RaspberryPi boards ([example](https://www.adafruit.com/product/1988?gclid=CjwKCAjwpqCZBhAbEiwAa7pXedAM0J-4kSNZ4Qz_9-5aFkVpvisTtj8U3HjhOb1kd7mQpAmnaLdyOxoCDpkQAvD_BwE)).

Then you can have the EuroPi connector hat PCB layout match appropriate microcontroller pins to the EuroPi peripheral header pins (basically translate a given board to the default EuroPi pico layout). This can be fabricated PCB or a simple DIY perfboard with shrouded wires connecting the microcontroller header pins to the appropriate EuroPi board pins. 

This allows us to experiment with the EuroPi using different microcontroller boards and unlock hardware potential not offered by the pico!

chrisib, 2022-11-01:
I'm not sure if it's possible given the rest of the hardware in this project, but it would be awesome if there was a usable MicroSD card slot in the expander. Being able to turn EuroPi into a Radio Music clone would be really nice.

I haven't looked through all the schematics, so I'm not sure if there are any SPI pins available for connecting to an SD card module. But if you're looking for ideas to add to the expander this would be near the top of my wish list.

Regarding the buttons, is there any point in making them recessed so you need a pin or other too to reach them? I definitely like the smaller buttons for something as potentially dangerous in a live set as the reset, but a recessed button is even safer.

glaklandy, 2023-05-02:
I'm all for front-facing USB and additional buttons/knobs and i/o via 4HP expanders. :)

roryjamesallen, 2024-02-06:
The [USB Router](https://www.allensynthesis.co.uk/modules/usb-router.html) now provides front access USB, and reset and bootloader mode can both be forced over REPL, and thus over USB. The EuroPi-X is going to have front access USB so this extra module won't be required for it, but for the current EuroPi this is now solved

closed 2024-02-06 by roryjamesallen