Allen-Synthesis/EuroPi #16: [Hardware Issue] ADC Scaling loses a lot of resolution unnecessarily

roryjamesallen, 2021-12-19:
# Hardware Issue

**Describe the bug**  
The scaling that happens to the CV input before the Pico ADC will translate +12V into only around 2.64V, meaning only 80% of the ADC is being used, and that's assuming a 0-12V input which is quite unlikely.
In the event of using a more common range of 0-8V, the Pico would only see 1.76V at the ADC, meaning only 53% of the ADC is being used, and thus decreasing the resolution from a theoretical maximum of 1.9mV/step to 3.6mV/step.

roryjamesallen, 2021-12-19:
The resistor values can potentially be changed to allow a greater range, or even alternative resistor values be given to users to allow them to choose which CV input range they would most likely use.

For example:
22k resistors will allow 0-(~15V) (way above necessary/useful range)
27k resistors will allow 0-(~12V) (still above the range of most CV sources)
30k resistors will allow 0-(~10.5V) (almost perfectly uses the whole ADC for the 0-10V CV range)
39k resistors will allow 0-(~8.5V) (almost perfectly uses the whole ADC for the 0-8V CV range)

So from these calculations, values of 30k or 39k (all resistors are the same in any given build, this is two possible build options) will allow the user to either have a CV input range of 0-10V or 0-8V, in either case making use of almost the entire (95%) ADC range.

I will experiment in hardware to make sure no clipping does occur, even in the worst cases of resistor tolerances, with each suggested resistor value

senorblasto, 2022-01-07:
I know this is meant to be a easy to make a affordable project, but just want to mention that that the standard way to scale down analog input voltages for an ADC is by using an Obama.

A quad Obama would only add a few dollars to the cost and allow for significantly increased input accuracy

roryjamesallen, 2022-01-07:
I'm assuming you mean op-amps! There are op-amps in the CV input stage, it's actually a Mutable Instruments circuit, I'm guessing the confusion is because the [Prototype](https://github.com/roryjamesallen/EuroPi-Prototype) didn't have any op-amps, but don't worry this one does for inputs and outputs, so you can use the full 10V range!

noises-modular, 2022-06-30:
Any guidance for allowing bipolar voltages for CV in, such as +/- 5V?

roryjamesallen, 2022-07-01:
> Any guidance for allowing bipolar voltages for CV in, such as +/- 5V?

Bipolar is a tough one. Essentially what it would involve is adding an accurate +5V offset to any incoming CV, and then adjusting the calibration so that it "thinks" +10V is actually +5V, +5V is actually 0V, and 0V is actually -5V, and tells the user that. The software part of that would be fairly easy, but the difficult bit, at least without some pretty complex bodged in hardware, would be the 5V offset

redoxcode, 2022-11-11:
> The resistor values can potentially be changed to allow a greater range, or even alternative resistor values be given to users to allow them to choose which CV input range they would most likely [...]

What are the exact resistors that need to be replaced? 
Would you need to add an additional over voltage protection?
Of you change them to 39k resistors (0-~8.5V) will it still be save to feed 10V or even 12V to the cv input?

roryjamesallen, 2022-11-11:
> What are the exact resistors that need to be replaced? 
> Would you need to add an additional over voltage protection?
> Of you change them to 39k resistors (0-~8.5V) will it still be save to feed 10V or even 12V to the cv input?
> 

R21, R22, and R23 I believe.

No extra protection required, you'd still be able to send in any voltage above the limit, it would simply be clipped. As the MCP6002 is rail to rail and powered by +3.3v the Pico's GPIO is always protected

iandoesallthethings, 2023-01-11:
> Bipolar is a tough one. Essentially what it would involve is adding an accurate +5V offset to any incoming CV, and then adjusting the calibration so that it "thinks" +10V is actually +5V, +5V is [...]

How hard would bipolar output be? I'd love to be able to generate +/-5v offsets.

roryjamesallen, 2024-02-15:
A new hardware revision of the current EuroPi is very unlikely to ever be made, and the X will be using external ADCs with a more carefully chosen range, so this issue is now redundant