Allen-Synthesis/EuroPi #22: Initial setup of EuroPi module throws an error because calibration.txt is not present.

awonak, 2022-01-09:
# Bug Report

**Describe the bug**  

Initializing the EuroPi module and installing the firmware for the first time throws an error because `calibration.txt` does not exist.

**To Reproduce**  
Steps to reproduce the behavior:
1. Nuke the pico filesystem or start with a new pico
2. Copy europi.py and install ssd1306 library
3. run any script that imports europi.py

**Expected behavior**  

When running europi.py firmware for the first time, the firmware should not crash when `calibration.txt` is not present.

**Additional context**  

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 188, in <module>
  File "<stdin>", line 126, in __init__
  File "<stdin>", line 29, in get_input_calibration_data
OSError: [Errno 2] ENOENT
```

roryjamesallen, 2022-01-10:
I believe this has been fixed in a more recent version of the europi library by changing 'r' to 'a+' to allow the file to be created if not present. I've just had a go with a freshly nuked Pico and it doesn't seem to throw an error and it does generate the calibration.txt file when required by an external program (even if the europi.py program has not been run as a script yet)

awonak, 2022-01-11:
I'm also getting the following error after the first prompt for 1V:
```
MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 231, in <module>
  File "<stdin>", line 129, in read_duty
MemoryError: 
>>> 
```
This can be fixed by reducing

awonak, 2022-01-11:
I must have been an old version of `europi.py` because I'm unable to reproduce. Thanks for the explanation!

closed 2022-01-11 by roryjamesallen

awonak, 2022-01-14:
I got this error again and traced down the root cause. I was initializing my nuked pico by copying `europi.py` to the root directory and omitted creating the `lib` directory. When `lib` is not present, micropython throws an error trying to read/write to a folder that does not exist on the pico.

roryjamesallen, 2022-01-14:
Ahh that is a very good find. An edge case for sure but I'll look at writing something in that gives a useful error message if it thinks it's being run outside of the lib folder