Allen-Synthesis/EuroPi #32: [Bug Report] Max output voltage capped at 9.12v?

awonak, 2022-01-21:
# Bug Report
Testing the `Output` class, I am unable to get the output voltage above 9.12v.

**To Reproduce**  
Steps to reproduce the behavior:

From a MicroPython shell running on the pico, I am testing setting the cv1 output duty cycle to the maximum value and I am only seeing 9.12V max.

```python
> cv1.pin.duty_u16(65535)
```
On my oscilloscope I am reading 9.12v. 

**Expected behavior**  

I would expect to see a maximum output of 10v.

roryjamesallen, 2022-01-21:
Ooer, this is pretty serious. I think it must be due to resistor combinations in the amplification stage for the outputs, but I'm not quite sure how it's losing so much voltage.
The calculation for  a non-inverting op-amp is 
Gain = 1 + Rf/Rin
Gain = 1 + 220k/100k
Gain = 1 + 2.2
Gain = 3.2

So in theory, with 3.3V output from the Pico, it should amplify to (3.3 * 3.2) = 10.56V, leaving a nice amount of headroom
I've measured the voltages at various points along the output stage and got 3.25V after the RC filter stage, and then 9.00V at the output pin of the op-amp, so somehow it clearly isn't being amplified to the factor that it should be

roryjamesallen, 2022-01-21:
Ok so lots of experimentation later, I have found that the problem is caused by the LED resistor forming a voltage divider with the internal resistor of the op-amp, and causing the voltage drop.
The good news is that by swapping the BOM to use 4.7k resistors instead of 1k for the LEDs, the output voltage is back above 10.3V (10V + a nice amount of headroom), and the LED is still nice and bright

mjaskula, 2022-01-21:
a couple things on this change:

- the schematic pdf (and I assume the kicad file) need to be updated with the resistor change
- the bill of materials should include R18 as a 5k1 resistor. 
- an update to or at least a note on the strip board layout.

roryjamesallen, 2022-01-22:
All edits have been made to the mentioned files to change the resistors to 4k7. I haven't got any on hand to double check but I've tested with 5.1k and it works perfectly, so if you've tested it and it's all good with 4k7 I think this is solved?

mjaskula, 2022-01-22:
We discussed the testing in discord. On my module, the 4k7 resistors yield a 10.4v output on the 10v output using the diagnostic program. This is with the default calibration.

closed 2022-01-22 by mjaskula

mjaskula, 2022-01-22:
Actually with the new default calibration, the 10v output is 10.07v

awonak, 2022-01-22:
Throwing my confirmation in here too. I just replaced R13 with a 5.1k resistor and I'm able to output a max voltage of 10.17v.

callimero, 2022-10-15:
BTW: Guess which Output I used for my breadboard build?

[image]