Allen-Synthesis/EuroPi #37: [Project Suggestion] Program selection menu

mjaskula, 2022-01-21:
# Project Suggestion: Program selection menu

As we add user contributed scripts we will (hopefully) end up with a large enough collection that the module will be useful 'out of the box' and not *require* that the user write their own scripts. In this case it would be nice to be able to switch amongst these scripts without connecting a computer and re-writing `main.py`.

I think that a proper menu system will require that we design two major pieces:
- hardware UX: the actual menu that the user experiences via the OLED, buttons, and knobs
- software API: The menu program and API that allows programs (apps?) to be included in the menu easily.

I believe that some work had already be done towards this, at least with the prototype module (https://github.com/awonak/EuroPiAlt). I suggest that we use this issue to focus these efforts and gather input on what this menu should be and how it should behave.

mjaskula, 2022-01-21:
Here are some of my ideas to get a conversation started

Hardware UX:
- should work on the base EuroPi and not require an expander, but it's ok if the expander makes the menu easier to use
- first boot should dump you to the menu, subsequent boots should run the last running program
- it should be easy to exit a program and return to the menu, but it should require purposeful action, the user shouldn't end up there by accident
- My first though on how it should behave: 
    - the screen shows a list of programs with one selected
    - the buttons are 'up' and 'down', allowing you to scroll through the list to select the program you want
    - a long press on both buttons launches the selected app
    - a long press on both buttons exits the app back to the menu

software API:
- Each program is an implementation of a base class which allows the user to define various aspects of the program including:
    - name
    - main loop
    - handlers
- We'll need a way of registering the available programs in a way that keeps the menu consistent between runs
- handlers should be registered on launch and removed on exit
- it would be nice to include a way for programs to save and load state

roryjamesallen, 2022-01-22:
I really like the bootloader idea and I'm sure I would use it on one of the EuroPis in my rack, but I think there are a lot of people (myself included in some cases) who would see the bootup taking you to a menu as a very bad thing, when one of the draws of modular is the immediacy of being ready to patch.
What would you guys think about developing the bootloader alongside the main branch as an alternate programming process, and then users can always swap between the two but there's still a clean and clear guide for one program, loads immediately?

I'm conscious that there have been design changes made, such as potentiometers instead of encoders, that are gearing the module towards standalone programs, and that's how I've been writing them (so that they are almost worthy of being made into a dedicated module). I kind of want to avoid the most common use of the module being another disting/O_c where people aren't as encouraged to write their own programs because it's seen from the surface as being a menu based multi-function!

mjaskula, 2022-01-22:
I definitely want the menu to be something that you visit intentionally. That's why I said:

> first boot should dump you to the menu, **subsequent boots should run the last running program**

I think the value of the menu at first boot is that the user can jump into using the module right away after getting it racked. They don't _need_ to write their own program. My assumption is that if they bought this thing, they _want_ to write their own, but not necessarily before they start wiggling.

That said, I think that the menu system would just be another program that the user has to load, coupled with a directory of the contributed programs. So the documentation could, remain as it is: (1) build, (2) optional calibration, (3) make it your own. And off to the side would be separate instructions to either (a) load a single program from the contrib dir, or (b) load the menu system along with the full contrib dir. So the question is how far off to the side does this need to be?

Also, I'll admit that as I've thought about the UI for different potential programs I've wished for encoders. I've also looked at the uO_c layout and thought that it might be a good direction for the next hardware iteration of the EuroPi to target. So, I'm glad to hear your thoughts and vision on this to that I can temper my expectations. And that if I do want to go in that direction it might be on my own project or expansion.

roryjamesallen, 2022-01-22:
Ah, I completely missed you saying subsequent boots should be into the previous program, that does sound like a great solution for storing multiple programs then.

I think the O_c layout is great for what it does, but accomplishing the same with the Pico would require quite a lot of extra hardware (and thus HP unless we make it SMD) as the Pico has very few on board ADCs. 

I do think encoders have a lot of advantages, but the simple fact that very very few beloved dedicated modules have encoders, even for 'digital' parameters (e.g. the new Mutable Beads has a potentiometer for the density control despite it selecting discrete values) proves that people like the feel of a smooth, click-less knob, with at the very least the *perception* of continuous control. 
I did deliberate for a long time when designing the new hardware about encoders, it's probably one of the things I was hung up on for the longest, so I'll agree with all the points in favour of them, I just think that the specific direction I'm going with this project (a platform with which people can design programs that are worthy of being standalone modules) leans towards potentiometers!

Of course there's also nothing to prevent anyone from designing an alternative hardware that uses encoders instead, I might even design an expansion module that has an encoder so it can be used for menu navigation if that's the direction lots of people seem to want

mjaskula, 2022-01-22:
No worries, I think, in general, we are on the same page.

It's nice to hear about these design decisions. You might consider writing some of them up, as blog posts, or whatever. I think it would help this fledgling community understand where you're coming from and the work you've put in a little more.

And yeah, an encoders expansion would be great.

roryjamesallen, 2022-01-22:
That's a good idea! I'll write some stuff up, possibly using GitHub discussions, although I've never used them before...

I'm thinking of an expansion in maybe 4HP with one or two encoders and one or two push buttons, that way you could have a single button dedicated to entering the bootloader and then an encoder that chooses the current program, or even something more complex like the way O_c can run multiple programs at once where you could run multiple simple programs at once using the same inputs e.g. top row of CV outputs are a sample and hold, bottom row is a clock divider

awonak, 2022-01-22:
I've got lots of thoughts and opinions on this topic, but I'll start with a few key observations from my work with the EuroPi-Prototype and my [Bootloader](https://github.com/awonak/EuroPiAlt) firmware. I'm in favor of the simple `europi.py` + `main.py` firmware and script approach as the default, and perhaps some additional directory for alternative firmwares (e.g. bootloader).

### EuroPi *must* be beginner friendly first
I have noticed folks from a wide range of backgrounds getting really interested in the EuroPi, some using this to try Python for the first time. I think it's important to ensure whatever changes we make, still keep the project easily accessible to beginners.

### Thonny is the default editor and it sucks at copying multiple files
I'm not aware of a way to copy entire directories or multiple files at once from Thonny to the pico. As a result, each individual contrib script would have to be copied manually. Yuck. This is why I have moved my development process over to `rshell` so I can use cp or rsync to copy multiple files in one shot. As my firmware and script library grew, I started using reusable libraries, so the number of files grew even more. As complexity grows we'll inevitably move in this direction too, so we'll need to keep in mind a good solution for simplifying this.

### Beware of hardware expansions
Early on I build a crappy [expansion board](https://github.com/awonak/EuroPi-Prototype/tree/main/contrib/expander) to give me input and additional knob control. This was awesome in theory, but I realized if I used the expansion module in any of my scripts, no one else would be able to use it. If we add any code that depends on expansion modules, then anyone without the expansion module won't be able to use it. Then you can get in the trap of forking scripts for use with/without expansion modules or excluding portions of the community who don't have that module. It's a tricky area.

### The bootloader is awesome
On my EuroPiAlt firmware, I love being able to load up any of my finished scripts for any patch I create without the need to reconnect the EuroPi to my computer. I see huge value in a bootloader script for those who want to run it. This is definitely an area worth exploring while ensuring EuroPi is still beginner friendly.

mjaskula, 2022-01-24:
I agree with everything Adam is saying here, that all sounds like the right philosophy to have.

Regarding expansion boards, I think they should be out of scope for this menu program discussion. That said, we should definitely make it a policy that no code gets merged to this repo that _requires_ an expansion board. If there's ever an official expansion board, or something third-party that is common in the community, then we can explore adding _support_ for it. But as Adam said, we will need to be careful.

awonak, 2022-01-26:
One design decision we'll have to consider here is the use of async. In my version of the firmware, the bootloader ran inside an [async event loop](https://github.com/awonak/EuroPiAlt/blob/main/src/bootloader.py#L94-L99). This allowed us to create async tasks like led triggers, button handlers, and other parallel tasks to improve performance and simplify code in some instance (but more complex in others).

As a result of this decision, scripts were required to define an [async main method](https://github.com/awonak/EuroPiAlt/blob/main/src/scripts/coin_toss.py#L76). In order to run, they would have to be started by an [async event loop](https://github.com/awonak/EuroPiAlt/blob/main/src/scripts/coin_toss.py#L104-L108
). Additionally, scripts needed to be manually added to the [bootloader script list](https://github.com/awonak/EuroPiAlt/blob/main/src/bootloader.py#L46). This might be desired if we want users to control what scripts to include/exclude, but it does add a required step when creating a new bootloader compatible script.

For our bootloader design, I'd like to avoid imposing script requirements like this as much as possible. If any requirements still exist, we should make that as easy as possible for folks, such as providing a base class with default behavior or solid documentation. Also, we should strive to make bootloader scripts backwards compatible with non-bootloader default firmware.

awonak, 2022-02-05:
Another thought that might help simplify the requirements for making a compatible script for the script selection menu is to use `boot.py` as the bootloader to choose from a list of script files and copy the selected file to `/pyboard/main.py`

https://docs.micropython.org/en/v1.9.3/pyboard/pyboard/tutorial/script.html

mjaskula, 2022-02-10:
I'm gonna run with this idea and see how far I get.

mjaskula, 2022-03-16:
This was completed with PRs #82 and #101.

closed 2022-03-16 by mjaskula