Allen-Synthesis/EuroPi #50: [Project Suggestion] Create EuroPiError class

awonak, 2022-01-23:
# Project Suggestion

When the EuroPi isn't connected to a serial output and an exception is raised, it's not obvious what happened. 

**Describe your suggestion or idea**  

It would be helpful to have a EuroPi specific error class that displays the error message to OLED when an exception is thrown.

This can be useful to explicitly throw:

```python
# europi.py
def choice(self, values, samples=None):
    if not isinstance(values, list):
        raise EuroPiError(f"choice expects a list, got: {values}")
```
The expectation would be that the error message fits as much of the exception text onto the oled display as possible. This makes it clear an error has occurred and helps hint at what the error was.

A more tricky case is when there is an unexpected error thrown. If it's not caught, then the oled wont show the error message.

One way to address this is to have a main entry point into the script and wrap that in a try/except/finally block.  For example:

```python
# coin_toss.py
try:
  coin_toss = CoinToss()
  coin_toss.main()
except BaseException as err:
  raise EuroPiError(e)
```
**Images**  
If applicable, add screenshots or photos to help explain your problem.

mjaskula, 2022-01-24:
Putting exception messages on this tiny screen sounds problematic, and so does putting them in front of users that didn't necessarily write or work on the code. That said, we clearly need some solution here in the event of a crash.

This might be a good place to use error codes. Print `Error 42` on the screen and the user can look it up in the docs, or post it to the discord support to get help.

awonak, 2022-01-24:
+1 Error codes sound like a much better solution.

roryjamesallen, 2022-01-25:
Error codes sound like a great idea, I'll start having a look at current errors which would be assigned codes

roryjamesallen, 2022-08-12:
Going to close this one for now, and then we can reopen when we find an appropriate use case for error codes

closed 2022-08-12 by roryjamesallen