Allen-Synthesis/EuroPi #59: [Documentation Suggestion] We shouldn't provide flashnuke.uf2

mjaskula, 2022-01-28:
# Documentation Suggestion

As suggested by @awonak. We don't own or maintain flashnuke.uf2, so it shouldn't be in our repo.

Remove flashnuke.uf2 from the repo and instead provide a URL to the official source of the file, and update any references in our docs

roryjamesallen, 2022-01-29:
Yup, very good point.
The original source is Adafruit, the link is here:
https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython

I will update any file referencing to the file and instead point it to that address, and then remove the file

roryjamesallen, 2022-01-30:
Now removed from the firmware folder and references point to the Adafruit link

closed 2022-01-30 by roryjamesallen