Allen-Synthesis/EuroPi #61: [Project Suggestion] Add type hinting to firmware code

awonak, 2022-01-29:
# Project Suggestion

MicroPython supports static type checking ([at least up to Python version 3.6](https://github.com/micropython/micropython/issues/3727)). The EuroPi firmware code should add type hinting to help improve static analysis and documentation for IDEs that support it.

**Describe your suggestion or idea**  

By adding type hinting, this will make writing scripts easier by providing the expected type of parameters of the API.

There are no negative side effects for IDEs like Thonny that don't use it.

mjaskula, 2022-02-10:
Type hinting in micropython seems to be the wild west right now:

https://github.com/micropython/micropython-lib/issues/190

It looks like we would need to include a stubs library like https://pypi.org/project/micropython-typing/ if we wanted to support this. So I think that means we'l need to have a better handle on our packaging before we tackle this.

So I think this issue depends on #71

pcurry, 2022-10-13:
Has type hinting improved since Feb. ? Function hinting at least seems to be supported.

mjaskula, 2022-10-15:
I haven’t looked into it since the 18 release.

awonak, 2023-04-08:
I think this has been open and unaddressed long enough that we can close it. If there is a renewed desire to take this on, we can reopen this issue.

closed 2023-04-08 by awonak