Allen-Synthesis/EuroPi #63: [Bug Report] Allen Synthesis banner should be optimized for web

awonak, 2022-01-30:
# Bug Report

I noticed the banner on the Allen Synthesis order confirmation page is 10MB which causes significant delays (50 seconds when not cached) in loading the page. The image should be resized to the view dimensions on the website. 

**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  

page:
https://allensynthesis.square.site/store/status/<CONFIRMATION_CODE>/confirmation

Link to unoptimized banner file: https://allensynthesis.square.site/uploads/b/927185487825331d77a55900406e125776a9e27e353ead17e3df7a79b6afc647/banner_1639401480.png

**Expected behavior**  

The file size shouldn't exceed 1mb and can probably be optimized down to a few kilobytes.

roryjamesallen, 2022-01-31:
I think I've fixed this now, the new image is 1920px wide and loads much faster

awonak, 2022-01-31:
Confirmed, the banner is now 165kb and loads in ~300ms!