Allen-Synthesis/EuroPi #71: [Project Suggestion] Make EuroPi firmware pip installable

awonak, 2022-02-05:
# Project Suggestion

To further simplify the setup process for EuroPi users, create a upip distribution package for the EuroPi firmware.

This would allow Thonny users to "install" the EuroPi firmware using the same Tools > Manage Packages process to install a "EuroPi firmware" library.

One missing requirement is defining a version scheme for differentiating between releases.

Official documentation:
https://docs.micropython.org/en/latest/reference/packages.html#distribution-packages-package-management-and-deploying-applications

https://packaging.python.org/en/latest/guides/distributing-packages-using-setuptools/

Practical example:
https://github.com/stlehmann/micropython-ssd1306

roryjamesallen, 2022-02-07:
This would be amazing, I've never done this before but looking at the docs I probably should be able to, so I can take this one

awonak, 2022-02-10:
For versioning the distribution, there is a standard called Semantic Versioning. Have a look at this site for some inspiration on how to handle versioning the firmware. https://semver.org/

roryjamesallen, 2022-02-10:
Just read through that, should we really be 'releasing' a stable version quite soon given it is technically a production API now?

awonak, 2022-02-10:
I would suggest we keep the firmware in "beta" as I'm certain we'll be making more changes as more of the first batch users start to make scripts and provide feedback. By keeping the firmware version in beta, that helps communicate the expectation to the users that there may be changes and additions to the firmware, and allows us to safely do so. With the patch notes Matt is adding, we can bump the minor/hotfix version number and add patch notes for our changes with each release.

Perhaps start us off at 0.2.0 and we can document the recent changes as 0.1.0 leading into the first distributed version.

Once we feel the momentum of initial feedback changes has slowed down, we can decide to release the first stable version "1.0.0" and leave it unchanged for a bit. Feature releases after that will likely slow down a bit after that.

awonak, 2022-02-10:
To clarify, we can distribute a beta version with pip. Technically the ssd1306 library is beta and its version is "0.3" https://github.com/stlehmann/micropython-ssd1306/blob/master/setup.py#L10

roryjamesallen, 2022-02-11:
Keeping in beta sounds good, I'll carry on reading to make sure I understand the process!

closed 2022-03-10 by roryjamesallen