Allen-Synthesis/EuroPi #83: [Documentation Suggestion]

awonak, 2022-02-16:
If a script is left running for a long period of time, it can burn that screen into the oled and leave ghost images. We should update the docs with suggestions 1) do not leave the EuroPi screen on for long periods of time with with something on the display, and 2) Update programming instructions to either let the display "sleep" after a period of time or provide some interaction to clear the display at rest.

roryjamesallen, 2022-02-17:
This could potentially use the frame buffer, as we don't want to send it to sleep if it's been on with moving images as this wouldn't cause burn in.
Maybe each time something already intensive happens, such as calling .show(), the frame buffer could be checked against a copy made of itself a given amount of time in the past, so if the buffer hasn't changed for a given number of ticks then a warning is given to the user

roryjamesallen, 2022-03-26:
I think this one we should just document in the contrib instructions, and mention that having a program with an unchanging screen for a long time is unadvised. I think it would be a mistake to hard code in some kind of timeout given that it might break programs that need to run fast, and would use up processing power to keep track of the counter/constantly check the elapsed time

awonak, 2022-11-18:
Thanks for adding this to the docs!

closed 2022-11-18 by awonak