Allen-Synthesis/EuroPi #97: [Bug Report] Display.clear() causes confusion

mjaskula, 2022-03-12:
# Bug Report

**Describe the bug**  
The function `Display.clear()` calls `show()` internally, as such it is of limited usefulness and requires detailed documentation in order to describe when best to use it. It is not currently in use by and contrib scripts, only in `europi.py`.

We should remove the function entirely and document the `fill(0)` method of clearing the screen.

roryjamesallen, 2022-03-12:
I agree, it causes more trouble than it's worth given that it implies itself as an easier way to clear the screen, but will likely result in much slower programs as it secretly calls the very intensive .show()

I think removing it entirely and documenting .fill(0) is a good solution

awonak, 2022-03-15:
Fixed in #98

closed 2022-03-15 by awonak