Allen-Synthesis/EuroPi #115: [Project Suggestion] GitHub/Discord bot for announcing new releases

awonak, 2022-04-01:
I think it would be neat to add a hook into our workflow to have a Discord bot automatically announce when a new version of the EuroPi firmware or contrib gets published to PiPy.

Our workflows are defined here:
https://github.com/Allen-Synthesis/EuroPi/tree/main/.github/workflows

I'm sure something like this must exist already, so hopefully this should be a low-lift contribution!

bigggmoustache, 2022-04-02:
Turns out, you really don't have to do anything. Github and Discord have webhook functions built in already. xD

https://gist.github.com/jagrosh/5b1761213e33fc5b54ec7f6379034a22

awonak, 2022-04-07:
The default GitHub integration is good for logging certain classes of events, but unfortunately doesn't give us the granularity of picking a specific workflow to create a chat message off of. That would be good if we want to log certain GitHub events to a discord channel, but that is not in scope for this issue.

awonak, 2022-04-07:
I suggest next steps you can look into a custom Discord webhook to parse the GitHub JSON and create a message in the #announcements channel when the "Publish distribution to PyPI" workflow completes successfully.
https://github.com/Allen-Synthesis/EuroPi/blob/main/.github/workflows/publish_to_pypi.yml#L42

roryjamesallen, 2022-04-30:
I think that realistically each release worth announcing will be announced manually in the GitHub with a relevant comment about the new features, and anything that wouldn't be worth announcing can still be checked on the changelog. If anything a bot might just result in too many messages in the announcements channel for only minor changes.
I'm going to close this issue, but we can always re-open it if we ever do toy with the idea of a bot/automatic announcements in the future!

closed 2022-04-30 by roryjamesallen