Allen-Synthesis/EuroPi #117: [Bug Report]

t-schreibs, 2022-04-05:
# Bug Report

**Describe the bug**  
When in external clock mode, coin_toss is triggered on the falling slope, rather than the rise

**To Reproduce**  
Steps to reproduce the behavior:
1. Run coin_toss.py
2. Set it to external clock mode
3. Feed it a 50% duty square wave as a clock
4. The tosses are triggered on the falling slope rather than the rise

**Expected behavior**  
It's expected that the tosses would be triggered when the rising slope is first detected

**Solution**
The problem appears to be here:
```
def wait(self):
        """Pause script execution waiting for next quarter note in the clock cycle."""
        if self.internal_clock:
            while True:
                if ticks_diff(self._deadline, ticks_ms()) <= 0:
                    self._deadline = self.get_next_deadline()
                    return
        else:  # External clock
            # Loop until digital in goes high (clock pulse received).
            while not self.internal_clock:
                if din.value() != self._prev_clock:
                    # We've detected a new clock value.
                    self._prev_clock = 1 if self._prev_clock == 0 else 0
                    # If the previous value is 0 then we are seeing a high 
                    # value for the first time, break wait and return.
                    if self._prev_clock == 0:
                        return
```
The second to last line should be `if self._prev_clock == 1:` since the `self._prev_clock` value was just changed.

t-schreibs, 2022-04-05:
This may or may not be unexpected behavior - but triggering on the falling part of the pulse seems unusual.

awonak, 2022-04-05:
Thanks for catching this and for the detailed bug fix tips! Yeah, this script was one of the first scripts written and came in before we added the rising and falling handlers, which would handle this much better: https://allen-synthesis.github.io/EuroPi/generated/europi.html#europi.DigitalInput

If you like this script, there's an even better version in the works written by @Bridgee in PR #104. Stay tuned for that one!

In the mean time, feel free to submit a pull request if you've got a fix, or I can work on it in the following days.

t-schreibs, 2022-04-05:
Oh, very cool! I'm looking forward to that one. The EuroPi is such a great project.

I'm on my phone right now, but can probably get a PR together tonight or tomorrow with the fix.

commit d09a8986 (2022-04-08): Fixes #117 - ext. clock now triggers on rising value.
commit ee7e3157 (2022-04-08): Merge pull request #119 from t-schreibs/main

Fixes #117 - ext. clock now triggers on rising value.
closed 2022-04-08 by awonak