Allen-Synthesis/EuroPi #123: [Bug Report] Potential issue with din handling some clocks

gamecat69, 2022-04-12:
# Bug Report

**Describe the bug**  
Some clocks appear to be too short to trigger the din handler.

**To Reproduce**  
Send a short clock to din. Some clocks trigger the din handler, others do not.

**Expected behavior**  
It is expected that a clock input to din will always trigger the din.handler.

**Images**  
n/a

**Additional context**  
Tested so far:
- Gates / square LFO with 50% duty cycle: works OK
- Malekko Varigate 4+: works OK
- Shakmat Time Apprentice sending clocks: causes this issue
- Shakmat Time Apprentice sending gates: works OK

roryjamesallen, 2022-04-14:
I don't have a Time Apprentice, is there any chance that someone has both a time apprentice and an oscilloscope so that they could see the length of the pulse when sending clocks?

roryjamesallen, 2022-04-24:
Updates from Discord:
Clocks under 0.01s seem less reliably detected by the firmware, however this could still be dependent on the currently running script.

Increasing the system clock speed improves reliability (minimum clock pulse that can be reliably detected), and increasing the system clock speed to as high as 250MHz seems to have no disadvantages,  so I am leaning towards adding that line to the firmware as it is. It can be overridden in each script if the user needs a slower system clock for some reason, but otherwise it seems that this line of code is being copied by most contributors into the individual scripts

gamecat69, 2022-06-07:
After some testing: Clocks >= 0.007s are OK, shorter and there will potentially be problems.

roryjamesallen, 2022-07-18:
This has now been added in a list of limitations to [programming_instructions](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/programming_instructions.md#programming-limitations)

closed 2022-07-18 by roryjamesallen

gamecat69, 2022-10-11:
I do. But it’s a cheap scope (DSO150) and difficult to measure time accurately. I will have a go this weekend and let you know.

Kind Regards,
Nik
(Sent from a mobile device)

bretdennison, 2022-11-11:
Tried getting my Europi to work with Pamela's New Workout, but no luck.  I even tried increasing the gate duty cycle (Width), but no luck.  It works fine with my WMD Metron as the clock.