Allen-Synthesis/EuroPi #124: [Bug Report] scripts that import micropython cause test failures

mjaskula, 2022-04-13:
# Bug Report

**Describe the bug**  
The EuroPi library checks for the existence of the `micropython` module to determine if it is in a micropython environment (running on hardware) or a cpython environment (running the unit tests). This was possible because no script's tests used the micropython library. Recently a change was made to import all scripts included in the menu as a part of menu unit test. This ensures that the menu can load, as it imports all of the scripts. The end result of this is that scripts that are included in the menu cause this test to fail if they import the micropython module.

**To Reproduce**  
Steps to reproduce the behavior:
1. Add the following line to any script that is used by the menu:

```import micropython```
2. run the test suite

**Expected behavior**  
Scripts can import micropython and their tests still pass.

**Additional context**  
This issue was discovered by the CVecorder script. It currently has its use of micropython commented out. As a part of resolving this issue those 3 lines should be uncommented.

Didah, 2022-05-05:
It should be possible to use the following blurb instead of relying on the import trick.

```python
import sys
if sys.implementation.name == "micropython":
    TEST_ENV = False # We're in micropython, so we can assume access to real hardware
else:  # implementation.name is probably "cpython"
    TEST_ENV = True
```

mjaskula, 2022-05-13:
Thanks @Didah! This works just fine. I'll be adding it to the fix PR for this issue.

mjaskula, 2022-05-18:
This was fixed with the merge of #143..

closed 2022-05-18 by mjaskula