Allen-Synthesis/EuroPi #132: [Bug Report] Ghost Button presses

gamecat69, 2022-04-29:
# Bug Report

**Describe the bug**  
This could potentially be a problem with the hardware or perhaps the firmware button debounce method.
On some occasions a b(n).handler_falling interrupt can be triggered without a physical press of a button. However, it is quite difficult to reproduce and seems to happen mostly randomly. The bug was identified after logging was added to the CVecorder script.

**To Reproduce**  
In some cases this can be produced by starting the module with nothing connected, then connecting CV to ain followed by a clock into din. However, this problem does not happen all the time and appears to happen rather randomly.

**Expected behavior**  
A b(n).handler_falling interrupt should not be triggered unless a button is pressed and then released.

**Images**  
[image]

**Additional context**  
This bug has been identified so far only when using the CVecorder script - as a button press can trigger the deletion of recorded CV, so it is therefore obvious when it happens. However, I believe this issue could manifest in all scripts.

gamecat69, 2022-08-03:
Confirmed to appear in probapoly and consequencer too.
Placing an upper time limit on a long button press detection dramatically reduces the chance of a phantom button press like the snippet below (3000 is used as an upper limit in this example).

`if ticks_diff(ticks_ms(), b1.last_pressed()) >  500 and ticks_diff(ticks_ms(), b1.last_pressed()) < 3000:`

gamecat69, 2022-11-14:
> I noticed these as well working on my own script. I know there are pullup resistors on the europi board, but also enabling the internal pullups of the pi pico seams to resolve these issues for me. I think my header pins on the pico are not the best and it only happens on some eurorack PSUs.
> To test this you can enable the pullups in your script like this:
>     b1 = Button(4)
>     b1.pin=Pin(4, Pin.IN,Pin.PULL_UP)
>     b2 = Button(5)
>     b2.pin=Pin(5, Pin.IN,Pin.PULL_UP)
>
> or directly change this line in europi.py
>     self.pin = Pin(pin, Pin.IN)
> to
>     self.pin = Pin(pin, Pin.IN,Pin.PULL_UP)
> There are some other things I want to optimize in the DigitalReader class, that might resolve or at least reduce this issue as well.
> I will propose a PR for these when I got time.

This is great! How do you feel about raising a PR for a change to the europi.py firmware?Kind Regards,Nik(Sent from a mobile device)

gamecat69, 2022-12-22:
I can confirm the suggested code change seems to avoid the phantom button presses

gamecat69, 2022-12-23:
Unfortunately, this change prevents the ability to long-press both buttons to return to the menu.

redoxcode, 2023-01-23:
Is this still present? There was some minor fix to the logic in https://github.com/Allen-Synthesis/EuroPi/pull/212

redoxcode, 2023-01-23:
> 

> Unfortunately, this change prevents the ability to long-press both buttons to return to the menu.
I guess you lost the irq handler for the both action by redefining the buttons.
That handler gets set in bootloader.py.
You should try to change europi.py to include the PULL_UP.