Allen-Synthesis/EuroPi #136: [Question] Possible To Use A C++ Program?

jakemumu, 2022-05-02:
# Question

Would it be possible to use this kit to run a C++ program as a module? I'm looking for a way to get started porting some audio plugins I've written to modules. They have JUCE as a dependency in C++ -- is it possible to connect the audio inputs from this board into a C++ app on the PI, and then back to the output? 

Appreciate your time sharing your thoughts on it!

pixiemars, 2022-05-03:
I think an electrosmith daisy patch submodule or an elk pi (if you can get one) is more suited to ports of juce based plugins.

roryjamesallen, 2022-05-03:
I agree with @pixiemars, this probably isn't the project for you unfortunately. Although the hardware technically could run C++ programs, and even probably the specific ones you would like to run, all of the documentation and libraries are for Python, so you would have a very tough time making everything work.

If you are well versed in programming with hardware it might not be too difficult, as most of the library is based around abstracting processes to make programming easier for beginners, so you could manage this if you consider the module to be more of a 'HAT' or a 'shield', where it's simply breaking out the pins and adding some extra hardware, but the software is still up to you .

jakemumu, 2022-05-03:
Thank you both for you response! 

While I'm a proficient engineer on desktop and have experience with Arduino, the pin programming on a PI is new to me is why I asked. I'll look into both of these projects! If I have no dice perhaps I could help extending this one to have the C++ apps as plug and play too.

closed 2022-05-03 by jakemumu