Allen-Synthesis/EuroPi #144: [Project Suggestion] Automated code quality tooling

Didah, 2022-05-18:
# Automated code quality tooling

Hey EuroPi friends!

I've noticed there's no code style rules set up yet. Would you be interested in having that set up?

I'm thinking Black as the formatter (with a line length of say 120, since the default 88 is a bit restrictive?) and pylint and flake8 for linting.
In addition to formatting/linting the current codebase we could add the same tooling to github actions so everything stays consistent going forward.

Let me know what you think, if this seems like a good idea I can draft a PR

As you can see below, this PR would introduce a lot of stylistic changes initially but after that the mental effort put towards code style will drop significantly, which might be useful as the number of contributors goes up.

[image]

roryjamesallen, 2022-05-23:
This is definitely something I'm on board with! We've been talking about using Black for formatting, and potentially making it the standard, in the Discord if you aren't already in there, so it should be happening soon! Feel free to make a PR if you are familiar with this sort of thing, I'm certainly not but I know the other maintainers have used Black before

Didah, 2022-05-23:
Glad to hear it. Here's my stab at the setup: https://github.com/Allen-Synthesis/EuroPi/pull/148