Allen-Synthesis/EuroPi #156: Upgrade to MicroPython 19.1

mjaskula, 2022-06-19:
# Upgrade to MicroPython 19.1

MicroPython 19.1 has been released. We should make sure that the EuroPi software is compatible. And then update our documentation.

Test the following under 19.1:

- micropython install procedure
- All hardware (diagnostic script may be a good way)
- all contrib scripts
- anything referenced in the 19.1 change log that looks relevant

Update the following:

- MicroPython Install process to reference 19.1
- write documentation on upgrading 18 -> 19.1

roryjamesallen, 2022-10-13:
I haven't tested every single script yet but I've 2 EuroPis running 19.1 for a week or two now and haven't seen any issues in the install procedure or operation

redoxcode, 2022-11-11:
Same here 1.9.1 works fine and how to setup EuroPi with up to date firmware is documented. Closed?

mjaskula, 2022-11-15:
Yeah, I think we're pretty close to closing this. I just upgraded one of my EuroPis and it seems fine. During the upgrade process I noticed some doc changes that I wanted to make, so I opened PR #197. I think that should do it for this issue.

closed 2022-12-01 by awonak