Allen-Synthesis/EuroPi #194: [Project Suggestion] Create custom firmware image that enables drag and drop setup

redoxcode, 2022-11-11:
# Project Suggestion

I suggest to add a custom firmware image (.uf2) file to the repo that allows users to setup a fresh pi pico just by connecting it to usb while the bootsel is pressed and drag-drop the custom firmware image on the device.
I think this would improve the experience for non tech-savvy people and increase the audience for the EuroPi project.
Also it would speed up the programming step for build-workshops or small scale production.

I created such a firmware as proof of concept and I will document the steps needed here soon.
The question is, if that could be automated or if it needs to be done manually for every new major release.

awonak, 2022-11-27:
This is awesome! Thanks for adding the details here and the link to your repo documentation. I was able to successfully follow the steps and compile my own uf2 firmware file. I've added some notes from my experiments to help automate this in a github workflow.

Workflow tips:

copy europi files
```shell
cp -r ~/Projects/EuroPi/software/firmware/*.py ./
mkdir experimental
cp -r ~/Projects/EuroPi/software/firmware/experimental ./
mkdir contrib
cp  ~/Projects/EuroPi/software/contrib/*.py ./contrib
```
fix memory size
```shell
sed -i 's/progsize=256)/progsize=1024)/g' _boot.py
```
create main.py
```shell
cat <<EOF >> main.py
import gc
gc.collect()
from contrib.menu import *
BootloaderMenu(EUROPI_SCRIPT_CLASSES).main()
EOF
```
get the ssd1306 lib
```shell
wget https://raw.githubusercontent.com/stlehmann/micropython-ssd1306/master/ssd1306.py
```

After all the module files have been copied and altered, call `make` from the rp2 directory

copy firmware.uf2 to versioned filename
```shell
grep -oP '(?<=")[\d\.]+(?=")' modules/version.py | xargs -I {} cp build-PICO/firmware.uf2 ./europi-v{}.uf2
```
Now you're all set! Copy the europi uf2 firmware to a freshly nuked pico and it should boot up.

awonak, 2022-11-27:
And now I've got a working GitHub action generating releases with attached `europi-vX.Y.X.uf2` attached when a new tag gets pushed :sunglasses: 

#198

closed 2022-12-01 by awonak