Allen-Synthesis/EuroPi #209: [Hardware Issue] Noise on knob values and analogue inputs

gamecat69, 2022-12-22:
# Hardware Issue

**Describe the bug**  
There is noise present when sampling the analogue inputs or knob values. This noise offsets the value read at the extremities of the readings, e.g 0 will never read 0 and the max value will never read the max value.

This issue causes the the percent() value to never read 100% on some modules. The issue also causes the analogue input to never read 0 Volts.

**To Reproduce**  
Steps to reproduce the behavior:
Read any value from k1 or ain to more than 1 decimal place several times.

**Expected behavior**  
It would be great if something like a deadzone was implemented. e.g. if really close to 0 the value returned is 0 or of close to the max the value returned is the max value.

gamecat69, 2022-12-22:
Note that a deadzone would be needed for both knobs and the analogue input. As the analogue input never truly reads the min or max value (in the same way as the knobs), due to the averaging done in _sample_adc().

gamecat69, 2022-12-23:
An alternative method would be to mask this problem at a lower level in the _sample_adc function using something like the example below. One benefit of doing this is that it would mask the problem in all higher level functions for both the knobs and the analogue input.

```
    def _sample_adc(self, samples=None, low_dead_zone=190, high_dead_zone=65500):
        # Over-samples the ADC and returns the average.
        values = []
        for _ in range(samples or self._samples):
            values.append(self.pin.read_u16())
        #return round(sum(values) / len(values))
        val = round(sum(values) / len(values))
        if val < low_dead_zone: # This is just noise, remove it by setting to 0
            return 0
        elif val > high_dead_zone: # probably noise, set to highest value
            return 65535
        else:
            return val
```

gamecat69, 2022-12-23:
I was a little concerned that in our haste to develop a fix, we perhaps had not correctly defined the problem, how to test for it and how to test that any change had fixed it lol! :)

... so I created a simple test script that displays the following:
1. Noise at the lowest and highest levels when reading from k1 and k2
2. Noise at the lowest level when reading from ain (I excluded the high val test for ain partially for speed and partially because I do not believe this will cause much of a problem for most)

When running the test, I believe the problem is present if any low value is greater than 0 and any high value is less than 1.0.

I ran this test twice, once using the suggested fix to _sample_adc and once using the suggested by @redoxcode.

The _sample_adc update fixed the problem for both the knobs and ain.
The suggestion by @redoxcode fixed the problem for the knobs, but not for the ain.

gamecat69, 2022-12-25:
> I don't think this should be a implemented in _sample_adc  as it returns an in from 0 to MAX_UINT16 that is the direct average of the samples that are in this range them self. Especially with sample size 1 this would result in a strange mapping. _sample_adc should do what everyone expects it to do.
> I haven't enabled deadzones for ain in my code since I don't think deadzones are the right way to handle this issue for ain. A change in way the calibration works would make more sense (not just calibrate the high point, but also the 0 point including any noise).
> But since my suggestion is implemented in the AnalogReader class it could be enabled for ain as well. But it shouldn't!
> I finished my tests on actual hardware and will make a PR later.

Good point around calibration for ain. I was thinking that might be a possibility. While calibration is mentioned, why don’t we include the knob deadzones in the calibrating routine too? But have default values for un calibrated modules like for the other values?Kind Regards,Nik(Sent from a mobile device)

gamecat69, 2023-03-27:
Has the analogue input noise issue been fixed or was a fix only developed
for k1 and K2? If the aim issue is still present, then I guess this issue
is still open.

redoxcode, 2023-03-27:
There was a PR that fixed 2 point calibration (https://github.com/Allen-Synthesis/EuroPi/pull/232)
idk, I don't have that issue so I can't test.

gamecat69, 2023-04-03:
@redoxcode: I am interested to hear what results you get when you run the test script I provided above.
Do you not experience the noise on ain? i.e. is the value 0 when nothing is plugged in when you run the test script?

redoxcode, 2023-04-05:
@gamecat69 I just made a fresh setup with the firmware 0.8.1 (not calibrated) and your [script](https://raw.githubusercontent.com/gamecat69/EuroPiApps/pot-test/software/contrib/noise_test.py) and I get 0.0 and 1.0 for both knobs and 0.0 for ain. I tested with and without usb power (always with eurorack power). Also I tested to have something constant connected to ain and that shows as well.

gamecat69, 2023-04-05:
Thanks for performing the test @redoxcode. I just did the same test on two modules using two different power supplies.
I got 0.0 and 1.0 values for k1 and k2, but non values > 0.0 for the ain test with nothing connected. I wonder where the ain noise is coming from in my setup....

redoxcode, 2023-04-05:
@gamecat69 can  you try to do the 2 point calibration and see if that helps?
Also maybe reflow the solder on the 3 pins of the socket?
Is the connection to ground working when nothing is connected?

> There was a PR that fixed 2 point calibration (#232)

redoxcode, 2023-08-15:
Closed? As OP hasn't answered and https://github.com/Allen-Synthesis/EuroPi/issues/211#issuecomment-1671746056 is closed as well?

gamecat69, 2023-08-15:
> Closed? As OP hasn't answered and #211 (comment)
> is closed as well?

Sorry for the delayed replies. I will test this with the latest firmware
and report back.

Kind Regards,
Nik

gamecat69, 2023-08-15:
I have done some testing, but TBH I got lost in PRs and sorta duplicate
issues in the repo :(.

I think that Issue #209 (
https://github.com/Allen-Synthesis/EuroPi/issues/209) should still be open
(see test results below).
However, the other PRs and issues etc that branched off from the original
issue may possibly be closed, but I guess this might be open to
interpretation.

Testing (issue 209)
I calibrated a module running firmware 0.10 and micropython 0.19. I used
the high-precision calibration routine.
My test script (
https://raw.githubusercontent.com/gamecat69/EuroPiApps/pot-test/software/contrib/noise_test.py)
resulted in a pass for the knobs and a very small amount of noise on the
ain (0.000562). I believe for this issue to be resolved the ain should read
0.0 because nothing is connected it should not be providing a non-zero
value.

Kind Regards,
Nik

roryjamesallen, 2023-08-21:
> 

Out of interest, what happens if you use the same precision for the calibration? i.e. change `calibrate.py` to use 4096 samples rather than 256 for each analogue input read. It is good that we're so so close to 0V, and almost any use of ain will naturally round that down to 0 (6 decimal places would let ain select between far more values than the specification of the EuroPi states based on the bit depth of the ADC), so this small discrepancy could just be down to using a lower number of samples when determining the zero point

roryjamesallen, 2023-08-21:
> a very small amount of noise on the ain (0.000562).

I've just run your noise script on an uncalibrated europi and got exactly 0.0 on ain, haven't yet tried with a calibrated one

gamecat69, 2023-08-22:
I calibrated using 4096 samples and ran the noise_test.py script again.
The results were interesting.... If I calibrate using 4096 samples I get
higher noise (0.03nnn). If I calibrate using 256 samples I get lower noise
(0.00002n).
I also did the test using different cases and I learned that the noise is
slightly different between cases... I therefore suspect that the noise in
my setup might be coming from the power rail, either from the power supply
or from other modules on the same rail....

Either way though.... my original suggestion was for the firmware to make
sure ain reads 0.0 even if the actual value reads as low as 0.00002), thus
eliminating any noise.
If this is not desired though, it is not a major problem. I will simply
continue to make sure that any scripts attempting to detect 0 on ain do
some rounding and the issue can be closed.

Kind Regards,
Nik

roryjamesallen, 2023-08-22:
> I get higher noise (0.03nnn). If I calibrate using 256 samples I get lower noise (0.00002n).

That is interesting, it does lean towards some kind of spiked noise then rather than a consistent noise floor, and that would explain more samples having higher noise as it picks up more of the spikes.
Either way, I think that this level of noise is acceptable as noise will be different in every system like you say, so any solution would be imperfect: if we find the average user noise at 256 samples and introduce firmware rounding to make sure that's zero, users with noisy cases will still experience noise at zero, and users with clean cases will have every ain reading slowed down by rounding unnecessarily

roryjamesallen, 2024-01-31:
With no realistic solution possible for this issue which doesn't sacrifice the experience of users with lower noise cases, and with the EuroPi-X development in progress which will fix these noise issues using higher quality ADCs and filtering, I'm going to mark this one closed unless any significant findings or solutions are found.
The Pico ADCs are very noisy, and from what I've read this in large part due to limitations of the ADC itself. Maybe in future this can be "fixed" with config menu settings to raise the voltage level that's considered as zero to account for a noise floor, but it might just be something we have to live with and accept that the hardware of the EuroPi is imperfect

closed 2024-01-31 by roryjamesallen