Allen-Synthesis/EuroPi #214: ADC problem

georgestheking, 2023-01-03:
The PICO is buggy and has limitation following PICO HARDWARE DATASHEET.
One idea will be to use an external ADC like the MCP3204.

Best regards

Georges

redoxcode, 2023-01-11:
Can you elaborate on how 'the PICO is buggy'?

georgestheking, 2023-01-11:
Hi,
A small problem for the ADC :
https://www.hackster.io/news/raspberry-pi-confirms-it-is-investigating-a-flaw-in-the-raspberry-pi-pico-rp2040-adc-95c393b55dfb

Best regards
Georges

roryjamesallen, 2023-02-12:
I understand the limitations with the inbuilt ADC of the Pico, but it is really too late to fix this now with the current hardware! Future hardware will almost certainly have external ADCs to allow there to be more of them as well, so I'll mark this as closed but know that it's on the list for future hardware improvements!

closed 2023-02-12 by roryjamesallen