Allen-Synthesis/EuroPi #215: [Question] limitations of Python script

gilbertohasnofb, 2023-01-03:
# Question

Hi, I’m interested in getting an EuroPi but I have some questions about it. In particular:

1. is there a size limit for a Python script to be suitable for EuroPi?
2. must the whole Python script be written as a single file or can it use local modules and packages?
3. can the script make use of locally saved files, e.g. a `.csv` file?
4. does it support scripts that use Python's Standard Library?

Many thanks in advance!

awonak, 2023-01-04:
> 1. is there a size limit for a Python script to be suitable for EuroPi?

The Pico has 264kB of SRAM, and 2MB of on-board flash memory, however Micropython leaves you with about 1MB flash memory. The EuroPi + ssd1306 libraries takes about 30k, so those are your memory constraints.

> 2. must the whole Python script be written as a single file or can it use local modules and packages?

The firmware uses MicroPython which behaves just like Python and allows importing of packages.

> 3. can the script make use of locally saved files, e.g. a .csv file?

Yep! We use this functionality in our  Save/Load State functionality [[docs](https://allen-synthesis.github.io/EuroPi/generated/europi_script.html#) | [code](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/firmware/europi_script.py)]

> does it support scripts that use Python's Standard Library?

Yes, anything that [MicroPython](https://micropython.org/) supports can be used in your script. 

> MicroPython is a lean and efficient implementation of the [Python 3](http://www.python.org/) programming language that includes a small subset of the Python standard library and is optimised to run on microcontrollers and in constrained environments.

I would suggest taking a look at the user contributed scripts to get a sense of what can be done with the EuroPi: https://github.com/Allen-Synthesis/EuroPi/tree/main/software/contrib

gilbertohasnofb, 2023-01-04:
@awonak Thank you so much for your detailed answer, this is super useful! I really appreciate it.

closed 2023-01-23 by gilbertohasnofb