Allen-Synthesis/EuroPi #217: [Documentation Suggestion] better menu doc to assist new users

mjaskula, 2023-02-04:
As [suggested by chrisib in discord](https://discord.com/channels/931297838804127794/931491803666993202/1071533554783244389):
> having a "here's how to use the menu" section in https://github.com/Allen-Synthesis/EuroPi/blob/main/software/programming_instructions.md would be nice? they specifically call out the contrib package and have a section dedicated to saving menu.py as main.py, but there's no explanation of what buttons and knobs to use to do anything.
> similarly, the menu doesn't seem to be documented at all on the main page of https://github.com/Allen-Synthesis/EuroPi/tree/main/software/contrib

chrisib, 2023-02-04:
https://github.com/Allen-Synthesis/EuroPi/pull/219

chrisib, 2023-04-08:
This can probably be closed now since the relevant changes have been merged.

closed 2023-04-08 by awonak