Allen-Synthesis/EuroPi #228: [Bug Report] Unable to calibrate

Salvadi, 2023-02-19:
I'm unable to calibrate the Europi with mode 1. The calibration gets stuck at "Calibrating..." after plugging the 10V source into the analogue input. 
The Europi was loaded with the the precompiled firmware from the "Quick start" section.

However programming the Europi with Thonny IDE I can calibrate with no problems at all.

**To Reproduce**  
Steps to reproduce the behavior:
1. Program the Europi as per the "Quick start" section 
2. Click on 'Calibrate'
3. Select 'Mode 1'
4. Get stuck at 'Calibrating...'

**Expected behavior**  
Get to the 'Insert analogue input into CV1' calibration step.

redoxcode, 2023-02-20:
Working on a fix here: https://github.com/Allen-Synthesis/EuroPi/pull/229

awonak, 2023-02-26:
Fixed in #229

closed 2023-02-26 by awonak