Allen-Synthesis/EuroPi #273: Pico W?

toneburst, 2023-05-08:
More of a question than an issue really: is the RPi Pico W able to be used as a drop-in replacement for the non-W variant?

Obviously it would use more power, if so…

toneburst, 2023-05-14:
Cool. Let me know how it goes!

I haven’t checked, but my suspicion is that the power requirements of the W variant may be significantly higher.

May or may not be a problem for the module’s transformer.

roryjamesallen, 2023-06-03:
> More of a question than an issue really: is the RPi Pico W able to be used as a drop-in replacement for the non-W variant?
> 
> Obviously it would use more power, if so…

Yes, the Pico W can be used very safely as a drop in replacement for a normal Pico!

Although it might draw slightly more, the 5V regulator on the EuroPi is really an overkill solution, so even though the entire module only draws around 90mA from the +12V rail, the regulator is capable of more than *>500mA* so if the W really needs more it will still be available to it

closed 2023-06-03 by roryjamesallen