Allen-Synthesis/EuroPi #381: Unable to install ssd1306 library

sumotoy, 2024-10-18:
# Bug Report
micropython-ssd1306 fail to install

Just followed the tutorial point by point on a fresh Win11.

After Thonny installation, tried to install as requested by tutorial the "micropython-ssd1306" library but...

 Installation fails. Get this error:

"gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Process returned with code 1"
Other libraries install correctly, but impossible to proceed without ssd lib!

[image]

sumotoy, 2024-10-18:
Fixed, for some reason I had to run Thonny as Admin, doing this, fixed the error!

closed 2024-10-18 by sumotoy

chrisib, 2024-10-18:
Glad you were able to sort it out. I haven't used Windows 11 and Thonny myself, so I don't know if this is a new issue. Might be worth checking with the Thonny developers to see if there's something that can be done to work around that.