Allen-Synthesis/EuroPi #394: Stabilize CV input with additional 1nF capacitor?

chrisib, 2024-11-24:
# Hardware Issue

A user reported on Discord that adding a 1uF capacitor in parallel with R23 improves the accuracy of the ADC, resulting in more accurate calibration.

Proposed modified input stage, adding `C*`:
[image]

This modification appears to match the original input stage, for example the one found in Braids:
[image]

# Planned Experiment

I've ordered some extra capacitors (I don't have any 1nF caps on-hand right now) and plan on modifying one of my EuroPis to include the proposed change.  Then I'll send in fixed voltages to AIN and record the raw PWM readings over some period of time, and compare the variance before & after the modification to see how much of a difference this really makes.

chrisib, 2024-11-24:
Started recording the data. I'm still waiting for the capacitors (I only ordered them yesterday, but they should arrive this week).

https://docs.google.com/spreadsheets/d/1rbvg_A5ZrKb3OqOu_JKJJ7-fPOTPx5xrRCnM5VENmN4/edit?usp=sharing

Script to generate the source data:
```python
from europi import ain, b1, oled
from time import sleep

def main():
    def wait_for_b1(value):
        while b1.value() != value:
            sleep(0.05)

    for volts in range(11):
        oled.centre_text(f'Connect {volts}V\nPress B1')
        wait_for_b1(1)

        with open(f'{volts}_v.csv', 'w') as fout:
            fout.write(f'{volts}V\n')
            for i in range(1000):
                duty = ain.pin.read_u16()
                fout.write(f'{duty}\n')
                
        wait_for_b1(0)
            
    oled.centre_text('Done')

if __name__ == '__main__':
    main()
```

civerachb-cpr, 2024-11-27:
Got my extra capacitors and added the 1nF to the back-side of the board, connected directly to the ends of R23.  Re-ran the code above. Results added to https://docs.google.com/spreadsheets/d/1rbvg_A5ZrKb3OqOu_JKJJ7-fPOTPx5xrRCnM5VENmN4/edit?usp=sharing

Interestingly at 0V the noise in the duty cycle actually seemed to go up. Could be a random event; we're not dealing with a lot of trials, so data's going to be subject to some variance.  But for all values above 0V, the standard deviation dropped by about half, which indicates much less noise in the duty cycle.

I'm tentatively willing to call the experiment a success, though the extra noise at 0V is worth investigating further.

If anyone else wants to repeat this experiment and post their results, I'd love to see the results of more data across a wider range of modules.

chrisib, 2024-11-28:
Got my extra capacitors and added the 1nF to the back-side of the board, connected directly to the ends of R23. Re-ran the code above. Results added to https://docs.google.com/spreadsheets/d/1rbvg_A5ZrKb3OqOu_JKJJ7-fPOTPx5xrRCnM5VENmN4/edit?usp=sharing

Interestingly at 0V the noise in the duty cycle actually seemed to go up. Could be a random event; we're not dealing with a lot of trials, so data's going to be subject to some variance. But for all values above 0V, the standard deviation dropped by about half, which indicates much less noise in the duty cycle.

I'm tentatively willing to call the experiment a success, though the extra noise at 0V is worth investigating further.

If anyone else wants to repeat this experiment and post their results, I'd love to see the results of more data across a wider range of modules.

chrisib, 2024-11-28:
Photos showing the 1nF capacitor soldered directly to the legs of R23; this is easiest if you have not cut the legs on the resistor too short.

[image]

[image]