Allen-Synthesis/EuroPi #422: [Bug Report] "About" script is broken

WavyDavy1960, 2025-03-05:
# Bug Report

**Describe the bug**
"About" script is broken, and locks up EuroPi when you press button #1 or #2.
 And it does not display version information.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'About'
2. Click on 'button #1 or #2'
3. no version information is displayed.
4. EuroPi is disabled, no button press or knob turning will work, until EuroPi is powered off, and then on again.

**Expected behavior**
Pressing button #2 should display firmware version.
**Images**
If applicable, add screenshots or photos to help explain your problem.

**Additional context**
Add any other context about the problem here.

chrisib, 2025-03-05:
Can you please provide the contents of `last_crash.txt` (if it exists) as well as `config/EuroPiConfig.json` and `config/ExperimentalConfig.json`?

WavyDavy1960, 2025-03-06:
> chrisib

I installed the V0.18.0 firmware/scripts, using the method involving powering up the EuroPi module, while holding down the small white button on the back of the module, and then dragging and dropping the firmware into the folder.
I don’t know how to install individual modules, or retrieve the files you requested.

I would welcome a link to directions on how to install individual modules, and how to retrieve the requested documents.

I am also interested in how to retain the calibration information, so I don’t have to recalibrate, each time I update the firmware/scripts.

I enjoyed building and using the EuroPi modules.
Thanks much for your work on this project.
DA
VE

chrisib, 2025-03-06:
Ah, ok. Rather than using the quickstart method using the all-in-one UF2 method, it's often more reliable to use the Thonny method described [here](https://github.com/Allen-Synthesis/EuroPi/blob/main/software/programming_instructions.md#pico-setup).

1. Use flash-nuke like you did initially to wipe the Pico
2. Download and install the Micropython firmware (a UF2 file) from https://micropython.org/download/RPI_PICO/ (if you are using a Pico W, Pico 2, or Pico 2 W there are alternative UF2 files for those -- make sure to download the correct one)
3. Install Thonny on your computer
4. Connect the Pico to your computer via the USB cable
5. Use Thonny's package manager to install the `ssd1306`, `micropython-europi` and `micropython-europi-contrib` packages onto the Pico

WavyDavy1960, 2025-03-06:
Hi Chris,
I need advice about the following error…
I believe I have a clone Raspberry Pi. It only says Raspberry Pi ©️ 2020, on the circuit board.
I am using the latest version of Thony downloaded from their website yesterday.
Using a Mac Mini with OS 12.7.6 (21H1320)

After following all instructions, in the link provided, I get stuck at the step where I try to install the ssd1306.
I tried with “MicroPython (Raspberry Pi Picco) Board in FS mode @ /de/cu.usbmodem14101”
I also tried with “MicroPython (RP2040) Board in FS mode @ /de/cu.usbmodem14101”
Selected in the lower right corner of Thony.
I am using the latest version of Thony downloaded from their website yesterday.
Using a Mac Mini with OS 12.7.6 (21H1320)

After clicking install, and waiting for a while, I get an error message as follows:

install --progress-bar off micropython-ssd1306
Looking in indexes: http://127.0.0.1:36628
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-ssd1306/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-ssd1306/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-ssd1306/
127.0.0.1 - - [06/Mar/2025 11:56:13] "GET /micropython-ssd1306/ HTTP/1.1" 200 -
127.0.0.1 - - [06/Mar/2025 11:56:13] "GET /micropython-ssd1306/ HTTP/1.1" 200 -
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 58065)
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 586, in __init__
    super().__init__(request, client_address, server)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 595, in do_GET
    self._serve_distribution_page(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 613, in _serve_distribution_page
    self.wfile.write("<!DOCTYPE html><html><body>\n".encode(SERVER_ENCODING))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 826, in write
    self._sock.sendall(b)
ConnectionResetError: [Errno 54] Connection reset by peer
----------------------------------------
127.0.0.1 - - [06/Mar/2025 11:56:13] "GET /micropython-ssd1306/ HTTP/1.1" 200 -
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 58086)
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 586, in __init__
    super().__init__(request, client_address, server)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
[... cut ...] "))

127.0.0.1 - - [06/Mar/2025 11:58:05] "GET /micropython-ssd1306/micropython-ssd1306-0.3.tar.gz/ HTTP/1.1" 200 -
Error Command '['/Users/dave/Library/Caches/pipkin/workspaces/d64e98649188e5065d1f585e6f96486a/bin/python3', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'micropython-ssd1306', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.

chrisib, 2025-03-07:
That looks like it may be a bug with Thonny. What version are you using, and does your Mac Mini have working internet connection?

If that problem persists you might want to reach out to the Thonny developers, as that doesn't look like normal behaviour.

Regarding the Pico, if it's a 2020 board it's likely either a standard Pico or a clone thereof. Some users have reported QC issues with third-party Pico clones, so it's possible that if your board is also a clone that there are hard-to-diagnose hardware or quality-control problems. That seems unlikely if some scripts work fine, but it's not out of the realm of possibility.

WavyDavy1960, 2025-03-07:
Hi Chris,
Disregard my previous email from earlier today.
I want to report that I was successful at installing the firmware and scripts, after closing my memory hogging browser, and restarting my computer.
I then tested the _About script, and it worked as expected, displaying the correct version of installed scripts.

Thanks for your help, answering questions.
And thanks again for all your contributions to the EuroPi project 
Have a great day
DA
VE

> On Mar 5, 2025, at 8:27 PM, Chris I-B ***@***.***> wrote:
> 
> here <https://github.com/Allen-Synthesis/EuroPi/blob/main/software/programming_instructions.md#pico-setup>

chrisib, 2025-03-07:
Great! Glad it's working for you!

closed 2025-03-07 by chrisib