Allen-Synthesis/EuroPi #446: Encoder Error?

Cyantig, 2025-09-03:
# Hardware Issue

I am having problems with my left encoder only moving one step forward or one step back and I can't scroll in the menu. I am able to enter and exit programs. Any idea on what the problem could be? I did check my soldering job before posting. 

Kat

chrisib, 2025-09-03:
Sorry to year you're having problems with your module!

Did the encoder previously work correctly and then stop, or has it never worked correctly?

Can you please try connecting the module to your computer via the USB cable and open Thonny. In Thonny's MicroPython terminal at the bottom of the window, please try running the following commands:

```python
from europi import k1

# rotate the knob all the way to the left
k1.percent()

# rotate the knob so it's close to vertical
k1.percent()

# rotate the knob all the way to the right
k1.percent()
```

These should output `0.0` followed by `0.5` and finally `1.0` (plus or minus a little bit of error).

Then try running

```python

# rotate the knob all the way to the left
k1._sample_adc(samples=128)

# rotate the knob all the way to the right
k1._sample_adc(samples=128)
```

That should help provide some insight into what could be going wrong.

Cyantig, 2025-09-07:
MicroPython v1.25.0-dirty on 2025-07-01; Raspberry Pi Pico2 with RP2350
Type "help()" for more information.
>>> from europi import k1
>>> # rotate the knob all the way to the left
k1.percent()
0.0
>>> # rotate the knob so it's close to vertical
k1.percent()
0.5340312
>>> # rotate the knob all the way to the right
k1.percent()

1.0
>>> 
---------------------------------------------
>>> # rotate the knob all the way to the left
k1._sample_adc(samples=128)
65535
>>> # rotate the knob all the way to the right
k1._sample_adc(samples=128)
36
>>> 
---------------------------------------------

Here is the output of the commands that you requested.  I installed the firmware from the all-in-one GitHub release.

CT

chrisib, 2025-09-07:
That looks good to me.

One thing I just realized: when navigating the menu you should be using `k2`, not `k1`. Can you verify that you can scroll through the menu by rotating the knob on the right?

Cyantig, 2025-09-07:
The behavior I am seeing on the right encoder gives no response at all in the menu. The only action I am getting at all is one step forward and one step back on the left  encoder.

chrisib, 2025-09-07:
Well, that looks to be the problem. It seems like it's probably a hardware issue. Can you please:

1. Measure the resistance of the potentiometer in the same 3 positions (all the way to the left, centre, all the way to the right). On my sample I measure 9.59kΩ all the way to the left, 5.24kΩ in the centre and 1.2Ω all the way to the right. Basically 0-10k across the full range of the pot.
2. Make sure you don't have any shorts between the 3 legs of the pot.
3. Make sure you have a good connections between the 3.3V pin on the Pico to one side leg of the pot, ground and the other side leg of the pot, and a between the centre leg of the pot and GP28 on the Pico

Hopefully it's an easy fix and there's just a solder bridge somewhere in the circuit path. At worst the potentiometer may be faulty and need replacing.

Cyantig, 2025-09-08:
So it looks like the 3v3 and ground lines are ok. 

So pot k1 seems to be labeled B10K measuring out at 5K-0k from the wiper, k2 seems to have a broken wiper as I am not measuring anything at all on it but 5K across the whole thing. 

I am going to go ahead a replace them both with some 10k and hopefully this will resolve any issues going forward.

Are you able to provide me with a part number that I could look up at Digikey?  

CT

closed 2025-09-08 by chrisib